ahdinosaur/blinksy #37: How to add new effect?

HaoboGu, 2025-05-28:
I want to add `Aurora` effecto from wled, is there any doc for adding new effects?

HaoboGu, 2025-05-28:
thanks for the detailed information! I'll check it out soon

ahdinosaur, 2025-05-28:
> maybe unrelated: I found this project is licensed by EUPL, can I use it in a MIT licenced project?

Good question. As far as I understand, yes.

EUPL is _weak_ copyleft, so not viral. Only changes to the Blinksy code must be EUPL, but any code that links to Blinksy can be whatever, even proprietary.

From Wikipedia:

> The EUPL is interoperable, without any "[viral effect](https://en.wikipedia.org/wiki/Viral_license)", in the case of [static and dynamic linking](https://en.wikipedia.org/wiki/Linker_(computing)).

See also in [this document](https://interoperable-europe.ec.europa.eu/collection/eupl/matrix-eupl-compatible-open-source-licences) the section 'Discussion on "Linking"'.

If I'm misunderstanding this then I'll need to reconsider my license decision.

HaoboGu, 2025-05-28:
Thanks! I'm closing this since all my questions are answered :D

closed 2025-05-28 by HaoboGu