ahdinosaur/blinksy #41: A raster-focused API

ahdinosaur, 2025-05-31:
Okay, so thinking more about [this](https://www.reddit.com/r/WLED/comments/1kwkg6a/comment/muybblm/) and [this](https://www.reddit.com/r/WLED/comments/1kwkg6a/comment/mus2xqc/) comment on Reddit.

> would this allow us to make our own screen sizes and then feed live video to it?

And

> Nice, it'd be cool to be able to add 8bit scenes/characters.

My reply on Reddit:

> Hi, my intended use case is less about [rasters](https://en.wikipedia.org/wiki/Raster_graphics) (grids of pixels) and more about [shaders](https://en.wikipedia.org/wiki/Shader) (given a position in space, or a position on a shape, what is the color?). While most people do 3d by mapping a 2d raster over a 3d surface, I want to do native 3d.
>
> That being said, the LED drivers I wrote in Rust (with help from [`smart-leds`](https://github.com/smart-leds-rs/smart-leds)) can be used for any use case, so you could definitely implement what you want if you were willing and able to write some Rust code. I think there'd be a better high-level API for what you want, my high-level API currently in Blinksy is for my shader-like use case.

Now, let's say I did add a raster-focused API to Blinksy.

How could this work?

For the abstraction like a layout, you'd define an array of "raster screens", basically a set of W x H dimensions.

Then for the abstraction like a pattern, you'd provide a "raster frame" for each screen.

This seems easy enough to implement, but maybe helpful for those who want to either do 8-bit pixel animations or pass video to their LEDs.

I'm not sure how to unify a raster mindset with a shader mindset. I guess within your pattern code, you'd want to be able to operate on the grid shapes you have (which is possible) and have an easy way to convert a raster frame into your iterator of colors for that shape.

Will let my back brain think about this, will admit is not a priority for me, but am curious.

crazyscot, 2026-08-29:
While my prototype is working well in blocking mode, it is proving challenging in async mode (which is what I want to get going, so I can use it with a TCP stack). Desktop async mode also works well; only the combination of target and async is troubled.

With a configured panel size above about 4x4, not all pixels light up, groups of pixels are wrong, and sometimes pixels flicker. This smells like a driver timing issue to me. 

Have you encountered anything like this, @ahdinosaur? GPT suggested the async chunked transmit path was the problem, but setting a large RMT buffer size (10240) only shuffled the corruption around the panel. Right now I wonder if there might be a stack overflow corrupting the buffer.

My target is a GL-C-016WL-D, with a single 16x16 ws2812 panel. Excuse the rough photo and improvised diffuser, but this is supposed to be primary blue (0,0,255) across the whole panel:

[image]

ahdinosaur, 2026-09-01:
> Have you encountered anything like this, @ahdinosaur?

No, but I can't say everything is well-tested, I haven't been doing much LED stuff as of late, so is very possible I left a bug somewhere, sorry. 🐛