ahdinosaur/blinksy #43: How... how do I use this?

nomonkeynodeal, 2025-06-04:
I have been looking around, but feel like I must be missing something obvious.

I have a 1000 LED volumetric cube running on a Dig-Quad with WLED 16 alpha.

https://i.imgur.com/L1BwEJZ.mp4

ahdinosaur, 2025-06-05:
Hi @nomonkeynodeal, sorry if things aren't clear, thanks for reaching out to ask!

> I have a 1000 LED volumetric cube running on a Dig-Quad with WLED 16 alpha.

That's awesome, and will soon be the perfect use case for Blinksy. The problem is at the moment, I haven't yet implemented 3d layouts and patterns. ( https://github.com/ahdinosaur/blinksy/issues/6 ) Can I get back to you in a few weeks? It won't be hard to add, since the architecture is already there. I'm making my own 3d cube using 16 x 16 panels (so not volumetric, but only because I'm lazy, I think volumetric will look cooler).

For a preview of how this would work, see [a comment I made on Reddit](https://www.reddit.com/r/FastLED/comments/1kwkh1o/comment/mv0kjea/):

> Hi, yes this is a perfect use case for Blinksy! The grid example doesn't quite capture the intention behind what I want, your spiral mapping is more what I want to enable.
>
> Essentially, where in my 2d example I use the [`layout2d!`](https://docs.rs/blinksy/latest/blinksy/macro.layout2d.html) macro and define an array with only a [`Shape2d::Grid`](https://docs.rs/blinksy/latest/blinksy/layout/enum.Shape2d.html#variant.Grid), you'll want to define an array of [`Shape2d::Point`](https://docs.rs/blinksy/latest/blinksy/layout/enum.Shape2d.html#variant.Point), where each point is the (x, y) position of the pixel. (See [`Layout2d`](https://docs.rs/blinksy/latest/blinksy/layout/trait.Layout2d.html) or the blog post where I talk about the bounds of the 2d space.) This array also needs to be in the daisy-chain pixel order.
>
> But since you probably don't want to write pixel layout array from scratch, you can look under the hood of the macro, the [`Layout2d`](https://docs.rs/blinksy/latest/blinksy/layout/trait.Layout2d.html) trait, and implement that. You'll need to provide the total pixel count and a method that returns an iterator of shapes (in your case, points).

Oh also, I haven't yet made a board support crate for the QuinLED boards. I just got mine from a friend traveling to New Zealand, so can be done as soon as I have the time.

I also realized though, my current approach to the LED drivers is not going to work for doing multiple lines in parallel. So I'll need to think about that, in the meantime this will mean a slow FPS. I'll have the same problem with my 1536 LED cube, so I'm be motivated to fix this. Just need to let my back brain come up with a good solution first.

Hope that helps for now. Please continue asking any and all questions, and hopefully soon you'll be able to run 3d-native patterns (effects) on your LED volumetric cube.

ahdinosaur, 2025-06-05:
May I ask, how did you make that LED volumetric cube? Are there any resources online for doing this?

nomonkeynodeal, 2025-06-05:
https://www.amazon.com/gp/aw/d/B0CGCH393Z

I got eight of those. They require almost no soldering. I thought about the best way to put them together and even tried soldering one layer (there are ten layers to my cube, each with four boards) but ultimately concluded four towers and an LED map was the way to go.

It's pretty cool and I'm particularly fond of the PS Fire effect in WLED 16 alpha (nightly). But I have been looking for real 3D effects for a while. You can do some pseudo 3D effects, but nothing to what a cube like this should be capable of.

Anyway, you can summon me whenever. Or I'll probably check back next year!